jerryscript-project/jerryscript #5054: A problem about calling sort function with long array

ayuan0828, 2023-03-16:
**Version:** `JerryScript 2.4.0`

**Description**:

 When I run the following program, I have an interesting discovery that jerryscript can neither output "`right`" normally, just like hermes; At the same time, it will not report overflow errors like xs. I think it may have some problems.

**Test case**:

```public class Test {
a = [];
a.length = 0xfffff;
a.sort();
print("right");
```

**Output of jerryscript**:

```

```

**Output of hermes**:

```
right
```

**Output of xs**:

```
Error: stack overflow
```

aksdfauytv, 2023-03-17:
I think ```ecma_deref_ecma_string (prop_name_p);``` is missing at the end of the first loop in ```ecma_builtin_array_prototype_object_sort```

Ginni1110, 2023-03-17:
May I ask if there is a problem when judging whether the length of the array is valid?

Is it executing this part of the code and exiting the loop directly?

```js
 if (ECMA_IS_VALUE_ERROR (get_desc))
    {
      ecma_collection_free (array_index_props_p);
      ecma_deref_ecma_string (prop_name_p);
      return get_desc;
    }
```

zherczeg, 2023-03-24:
I checked the latest (1a2c04763aba49f52b1537acd3730098c873511c) jerry, and I got JERRY_FATAL_OUT_OF_MEMORY for this example. It fails when it copies the data into an internal sort buffer.