jerryscript-project/jerryscript #5057: Segmentation fault in jerry

marckwei, 2023-03-27:
First of all, when reporting a bug, give the issue a descriptive title.

In the body of the issue, optionally give a free-form text description of the
bug. Give the context necessary for others to understand the problem.

Then, provide information necessary to reproduce the bug.
Omit sections that are irrelevant for the bug report, but note that information
like git revision, build platform, build command, and test case are required in
almost all cases.

###### JerryScript revision
1a2c04763aba49f52b1537acd3730098c873511c

###### Build platform
Ubuntu 20.04.2 LTS (Linux 5.15.0-67-generic x86_64)

###### Build steps
./tools/build.py

###### Test case
```
var p = new Promise(function(resolve,reject){})
var f = async(p) =>{}

await p;

f(p).then(function(){})
```

###### Execution platform
same as the build platform.

###### Execution steps
```sh
build/bin/jerry testcase.js
```

###### Output
Sometimes throws syntax errors: Unhandled exception: SyntaxError
Most of the time jerry crashes: Segmentation fault (core dumped)

###### Backtrace

###### Expected behavior
not to crash

carnil, 2023-04-26:
CVE-2023-30408 was assigned for this issue.

happy-boy-maker, 2025-01-09:
针对这个issue，我发现在2.3与3.0版本都不会复现这个问题。目前看只有2.4会出现 jerry crashes: Segmentation fault (core dumped)。现在，jerry3.0.0可以正常识别到对应的语法错误，正常情况下await只能在async函数里。所以是否可以认为这个问题现在不存在，关闭该issue。如果我的理解有问题，欢迎指正。
[image]

happy-boy-maker, 2025-01-09:
For this issue, I have found that this issue does not reappear in versions 2.3 and 3.0. At present, only 2.4 jerry crashes will occur: Segmentation fault (core dumping). Now jerry3.0.0 will normally recognize the corresponding syntax error. Normally await can only be used in async functions. So can we consider that this problem does not exist now and close the issue? If I have a problem with my understanding, please correct me.
[image]