jerryscript-project/jerryscript #5058: Segmentation fault in jerry

marckwei, 2023-03-28:
###### JerryScript revision
1a2c04763aba49f52b1537acd3730098c873511c

###### Build platform
Ubuntu 20.04.2 LTS (Linux 5.15.0-67-generic x86_64)

###### Build steps
Describe how to build JerryScript. Give all the necessary details of the build
(e.g., environment variables, command(s), profile, command line options, etc.).

E.g.:
```sh
tools/build.py --compile-flag=-fsanitize=address --compile-flag=-g
```

###### Build log

###### Test case
```JavaScript
function i(a,b=eval()){eval()}
i(i("a"),eval("var a")) 
```

###### Execution platform
same as the build platform.

###### Execution steps
```sh
build/bin/jerry testcase.js
```

###### Output
Segmentation fault (core dumped)

###### Backtrace
AddressSanitizer:DEADLYSIGNAL
=================================================================
==167385==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000004 (pc 0x562a05bd8581 bp 0x7fffdc027800 sp 0x7fffdc027790 T0)
==167385==The signal is caused by a READ memory access.
==167385==Hint: address points to the zero page.
    #0 0x562a05bd8580  (/data/jerryscript/asan/bin/jerry+0xd0580)
    #1 0x562a05b8051c  (/data/jerryscript/asan/bin/jerry+0x7851c)
    #2 0x562a05b9069f  (/data/jerryscript/asan/bin/jerry+0x8869f)
    #3 0x562a05b919b9  (/data/jerryscript/asan/bin/jerry+0x899b9)
    #4 0x562a05b6b6d8  (/data/jerryscript/asan/bin/jerry+0x636d8)
    #5 0x562a05bc14cd  (/data/jerryscript/asan/bin/jerry+0xb94cd)
    #6 0x562a05bc696a  (/data/jerryscript/asan/bin/jerry+0xbe96a)
    #7 0x562a05b9085b  (/data/jerryscript/asan/bin/jerry+0x8885b)
    #8 0x562a05b919b9  (/data/jerryscript/asan/bin/jerry+0x899b9)
    #9 0x562a05bf8576  (/data/jerryscript/asan/bin/jerry+0xf0576)
    #10 0x562a05b25dac  (/data/jerryscript/asan/bin/jerry+0x1ddac)
    #11 0x7f32bb095082 in __libc_start_main ../csu/libc-start.c:308
    #12 0x562a05b26cfd  (/data/jerryscript/asan/bin/jerry+0x1ecfd)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/data/jerryscript/asan/bin/jerry+0xd0580) 
==167385==ABORTING

###### Expected behavior

aksdfauytv, 2023-04-03:
It doesn't crash for me. Did you give it enough RAM/stack?

marckwei, 2023-04-05:
To reproduce，you should use the same build args as me

marckwei, 2023-04-05:
> It doesn't crash for me. Did you give it enough RAM/stack?

To reproduce，you should use the same build args as me

aksdfauytv, 2023-04-05:
actually, if you build with debug enabled, you get an assert earlier:
```
ICE: Assertion 'ecma_get_lex_env_type (lex_env_p) == ECMA_LEXICAL_ENVIRONMENT_DECLARATIVE' failed at /home/maciek/jerryscript/jerry-core/vm/vm.c(vm_loop):1441.
```

carnil, 2023-04-26:
CVE-2023-30406 seems to have been assigned for this issue.

happy-boy-maker, 2025-01-09:
For this problem, I turned on debug. We do get 'ecma_get_lex_env_type (lex_env_p) == ECMA_LEXICAL_ENVIRONMENT_DECLARATIVE' failed. I look at the condition that this assertion is triggered by. It seems that he is looking for a value in the external lexical environment and will look for an attribute in the final lexical environment. Can any failure in this process be a default that the developer's code has a syntax error? I encountered a case where lex_env_p->u2.outer_reference_cp was finally empty on jerry2.3. The check seems to be caused by OOM (stack overflow). I wonder if anyone knows why this happens and how stack overflow affects the empty pointer. And I hope the official can give a clear reply whether this is a bug?