jerryscript-project/jerryscript #5077: Inconsistent behavior

Georgezxh, 2023-04-27:
### Description
Other js engines can print correctly while jerry throws a type error. 
### JerryScript revision
2.4.0

### Build platform
ubuntu20.04

### Test case
```
result = String.prototype.normalize.call(String.prototype.padStart.call(0, 5, '-'));
print(result);
```

### Output
Unhandled exception: TypeError

### Expected behavior
----0

matetokodi, 2023-06-28:
Currently `String.prototype.normalize` is not implemented because it would require the use of `libICU` or a similar library for proper unicode and locale support, which would increase the size of the jerryscript binary substantially, and as such it will not be done.

zherczeg, 2023-07-18:
Adding an icu would increase the library size too much.