jerryscript-project/jerryscript #5115: Not feasible to update some CI jobs to ubuntu 22.04

matetokodi, 2023-11-29:
The following CI jobs have dependencies that cannot be satisfied on more recent versions of ubuntu than 18.04:
 - MbedOS_K64F_Build_Test
    - Requires the `gcc-arm-embedded` toolchain from the `ppa:team-gcc-arm-embedded/ppa` ppa, which only has packages for ubuntu 14.04, 16.04, and 18.04.
    - Requires the `mbed-cli` which uses python features that were deprecated in version 3.3, and removed in 3.10 (ubuntu 22.04 ships python version 3.10)
    - Requires gcc arm toolchain version >= 9 and < 10 (ubuntu 22.04 ships versions 10 and 11)

 - RIOT_STM32F4_Build_Test
    - It was also previously using `gcc-arm-embedded` toolchain from the `ppa:team-gcc-arm-embedded/ppa` ppa, but it does successfully build using `gcc-arm-none-eabi` from the ubuntu repos on top of ubuntu-latest, however I'm unsure about this being the right compiler for this target
      If it is the right one, then it can be re-enabled with minimal changes to the `gh-actions.yml` configuration; however if it is not then it cannot be updated to a newer ubuntu version.

A possible workaround to make them work, since Github Actions dropped support for ubuntu 18.04 runners, is to use docker to run ubuntu 18.04 inside a container on top of ubuntu-latest: for as long as the ubuntu 18.04 repos and the docker image is available, they should continue to work, with some changes to the `gh-actions.yml` configuration.

Should we use this workaround, or is it not worth the trouble and added dependencies on docker images, and drop these tests?

zherczeg, 2023-11-29:
@akosthekiss what do you think about supporting these platforms?

akosthekiss, 2023-12-11:
Mbed OS will always have a special place in my heart. I'd love to see it stick around.
- According to the latest release notes (https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-6.17.0), it can be built using GCC_ARM 10.3-2021.07 and the K64F target is still supported.
- According to the tools doc of Mbed OS (https://os.mbed.com/docs/mbed-os/v6.16/build-tools/index.html), recent Mbed OS (>6.5) uses Mbed CLI 2 instead of the deprecated Mbed CLI 1.
- According to the docs of Mbed CLI 2 (https://os.mbed.com/docs/mbed-os/v6.16/build-tools/install-or-upgrade.html), the new python package name is mbed-tools (and it requires cmake and ninja as well).
- According to the tools doc, in sync with the release notes, the supported gcc version is GNU Arm Embedded version 10 (10.3-2021.07). However, the landing page of the GNU Arm Embedded Toolchain is deprecated, and there are newer releases available at https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads . It would be interesting to see whether these newer versions can also build Mbed OS.
- According to discussions on askUbuntu (https://askubuntu.com/questions/1243252/how-to-install-arm-none-eabi-gdb-on-ubuntu-20-04-lts-focal-fossa), there are no more ppa releases for gcc embedded. So, the toolchain may need to be downloaded and installed manually. (Best practices and scripts are shared in the discussion linked above.) But the Ubuntu-packaged version may also be worth trying.

In summary: it may be a bit more work to revive MbedOS_K64F_Build_Test, but should be doable.

As for RIOT, I don't have strong opinions. In general, if we can keep support for a target with minimal effort, then we should not drop it. Regarding the toolchain, according to my best knowledge, gcc-arm-embedded was actually gcc-arm-none-eabi. So, Ubuntu-packaged gcc-arm-none-eabi might just be fine. (But it it always worth consulting the project's documentation: https://doc.riot-os.org/getting-started.html . There it is stated that "GCC, binutils, and newlib for arm-none-eabi" are needed. So, same conclusion.)

closed 2023-12-15 by matetokodi