jerryscript-project/jerryscript #5124: request a new release

chenrui333, 2024-01-21:
👋 it has been couple of years since last release in 2021, would be good to make a new release including all the recent changes, thanks!

alex1891, 2024-10-22:
is here anybody still maintaining this project?

alex1891, 2024-10-23:
> The pace is not that fast nowadays, but the maintainers are here, yes.

"pace is not that fast" is good, no release since 2021 and a ton of vulnerabilities

akosthekiss, 2024-10-23:
Sarcasm. Duly noted.

Any contribution from anyone willing to spend their resources on fixing those issues faster is warmly welcome.

(BTW, Hacktoberfest is still on. Happy to put the hacktoberfest-accepted label on any quality PR.)

alex1891, 2024-10-23:
the thing is, the issues are fixed because of the greate OSS community - what is missing is a tag for a release :)

zherczeg, 2024-10-24:
Is there a product which need this release? Where do you want to use it?

alex1891, 2024-10-24:
> Is there a product which need this release? Where do you want to use it?

Yes, we are using it in an commerical HMI software.

alex1891, 2024-11-07:
so are there any plans for an release?

ossy-szeged, 2024-11-07:
> so are there any plans for a release?
Yes, we are planning it, hopefully we will manage to release 3.0.0 this year.

LaszloLango, 2024-12-18:
JerryScript 3.0 is officially released, https://github.com/jerryscript-project/jerryscript/releases/tag/v3.0.0

closed 2024-12-18 by LaszloLango