jerryscript-project/jerryscript #5128: How to ensure that arguments of c method are not destroyed after the execution of the function

swordsmanSun, 2024-02-11:
I can't persist the argument. It appears to be destroyed after the c method is executed, even though I used jerry value copy api

I want to persist the js callback argument received by the c native method and call it in another scope, but I can't access it properly in that other scope. The following is a simple example( function eventHandler is called afhter function initLoopEvent is called, and function eventHandler is called multiple times )

```c
static esp_err_t eventHandler(void* ctx, system_event_t* event) {
    jerry_value_t initLoopEventCallback = *(jerry_value_t*)ctx;

    if (jerry_value_is_function(initLoopEventCallback)) {
        // false
       //  The program will never get here
    } else {
        // true
        jerry_log(JERRY_LOG_LEVEL_ERROR, "initLoopEventCallback is not a function");
    }
}

static jerry_value_t initLoopEvent(const jerry_call_info_t* call_info_p,
    const jerry_value_t arguments[],
    const jerry_length_t arguments_count) {

    jerry_value_t callback = jerry_value_copy(arguments[0]);
    error = validateFunction(callback);
    if (isError(error)) {
        jerry_log(JERRY_LOG_LEVEL_ERROR, "Invalid arguments[0]");
        jerry_log(JERRY_LOG_LEVEL_ERROR, "Invalid function initLoopEvent");
        return error;
    }

    // init evemt loop
    ESP_ERROR_CHECK(esp_event_loop_init(eventHandler, &callback));

    return jerry_undefined();
}

```

zherczeg, 2024-02-11:
The `jerry_value_copy` is correct way to create a new reference. Maybe somewhere it is unintentionally released.

swordsmanSun, 2024-02-11:
> The `jerry_value_copy` is correct way to create a new reference. Maybe somewhere it is unintentionally released.

thanks，i will take a check

closed 2024-02-12 by swordsmanSun