jerryscript-project/jerryscript #5135: Assertion ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p) in ecma_free_string_list

Qbtly, 2024-03-26:
###### JerryScript revision
cefd391772529c8a9531d7b3c244d78d38be47c6

###### Build platform
Ubuntu 22.04.3

###### Build steps
```sh
python ./tools/build.py --builddir=xxx --debug --clean --compile-flag=-fsanitize=address --compile-flag=-g --strip=off --lto=off --logging=on --line-info=on --error-message=on --stack-limit=20
```

###### Test case
```sh
//poc1.js
class s { constructor ( ) { } [ { [idx1] : { [{ [idx1] : { [idx2] : {}}} = [5.7]] : {}}} = [5.7] ] = class { } } 
new s ( ) ; 
```
```sh
//poc2.js
const obj = { a : (o) = 1 };
var a = { a : (o) = 1 } = obj;

(class C2 { ["foo"] = 5; ["foo"] = {} + {} })((new (class o { ["foo"] = 5; [{ a : (o) = 1 } = obj] = {} + {} })).foo === 5);
print(o === 1);
```

###### Execution steps
```sh
./xxx/bin/jerry poc.js
```

###### Output
```sh
//poc1.js
Unhandled exception: ReferenceError: idx1 is not defined
0: /home/qbtly/Desktop/PatchFuzz/js/output/myh_jerry/jsc/crashes/i
ICE: Assertion 'ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)' failed at /home/qbtly/Desktop/target/jerryscript/jerry-core/ecma/base/ecma-literal-storage.c(ecma_free_string_list):75.
Error: JERRY_FATAL_FAILED_ASSERTION
```
```sh
//poc2.js
Unhandled exception: ReferenceError: Variables declared by let/const must be initial
0: /home/qbtly/Desktop/PatchFuzz/js/output/myh_jerry/jsc/crashes/i
ICE: Assertion 'ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)' failed at /home/qbtly/Desktop/target/jerryscript/jerry-core/ecma/base/ecma-literal-storage.c(ecma_free_string_list):75.
Error: JERRY_FATAL_FAILED_ASSERTION
```

lygstate, 2024-11-14:
does https://github.com/jerryscript-project/jerryscript/issues/4525 related to this?

happy-boy-maker, 2024-11-28:
So far, the preliminary conclusion in the comments section is that this is a memory leak? I think the last issue with a similar error is January 2021. Does the community have a corresponding plan to fix this problem? Or what are the current conclusions about this issue, such as the impact is relatively small?

akosthekiss, 2024-11-28:
@happy-boy-maker you a bot?

happy-boy-maker, 2024-11-28:
@akosthekiss I don't know what makes you think I'm a robot.

akosthekiss, 2024-11-28:
- You have summarised the above mentioned, potentially related issue for no apparent reason.
- Then ask for plans and conclusions, without giving context on why it is of importance to you.
- All this without any history of involvement with the project.

So, please, shed light on context.

happy-boy-maker, 2024-11-29:
@akosthekiss Our project uses Jerry as the JS engine. The issue released on the official platform can help us check whether some problems are introduced by ourselves. Memory leakage is a serious problem. So, I'd like to know how jerry's official attitude to this issue is. The triggering conditions provided by developers seem to be triggered by incorrect JS cases. Whether this problem does not affect the normal use of Jerry can be avoided by verifying the code.

LaszloLango, 2024-12-19:
> does #4525 related to this?

Sorry for the late response. Yes, probably #4525 is the minimized testcase for this fuzzer testcase.