jerryscript-project/jerryscript #5154: Abort in jerry

Qbtly, 2024-08-06:
###### Commit ID
2dbb6f79efa08767ce44fbc8e3f26e80649295d7

###### Build platform
Ubuntu 22.04.3

###### Build steps
```sh
python3 ./tools/build.py --builddir=xxx --clean --compile-flag=-fsanitize=address --compile-flag=-g --strip=off --lto=off --logging=on --line-info=on --error-message=on --stack-limit=20
```

###### Test case
```sh
function a(lastIndex) {
    let fake_re = {
      exec: () => Symbol,
  
      get lastIndex() {
        return lastIndex;
      },
  
      set lastIndex(value) {},
  
      get global() {
        return true;
      }
    };
  
    RegExp.prototype[Symbol.replace].call(fake_re, "");
  }
  
  a(0);
```

###### Execution steps
```sh
./jerry poc.js
```

###### Output
```sh
Release:
Program received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737350473600) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737350473600) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737350473600) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737350473600, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff7cc6476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff7cac7f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x000000000080777e in jerry_port_fatal (code=code@entry=JERRY_FATAL_REF_COUNT_LIMIT) at /jerryscript/jerry-port/common/jerry-port-process.c:29
#6  0x0000000000632ab7 in jerry_fatal (code=3616307, code@entry=JERRY_FATAL_REF_COUNT_LIMIT) at /jerryscript/jerry-core/jrt/jrt-fatals.c:63
#7  0x000000000050b978 in ecma_ref_object_inline (object_p=<optimized out>) at /jerryscript/jerry-core/ecma/base/ecma-gc.c:143
#8  0x000000000054c803 in ecma_copy_value (value=1019) at /jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:894
#9  0x000000000054c989 in ecma_fast_copy_value (value=3616307) at /jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:921
#10 0x00000000007864e2 in ecma_op_resolve_reference_value (lex_env_p=0x12212f8 <jerry_global_heap+248>, name_p=0x104d) at /jerryscript/jerry-core/ecma/operations/ecma-reference.c:300
#11 0x00000000006c6d4d in vm_loop (frame_ctx_p=0x7fffffffd080) at /jerryscript/jerry-core/vm/vm.c:1044
#12 0x00000000006c2c00 in vm_execute (frame_ctx_p=frame_ctx_p@entry=0x7fffffffd080) at /jerryscript/jerry-core/vm/vm.c:5211
#13 0x00000000006c0ce4 in vm_run (shared_p=shared_p@entry=0x7fffffffd1a0, this_binding_value=this_binding_value@entry=11, lex_env_p=lex_env_p@entry=0x12214e8 <jerry_global_heap+744>)
    at /jerryscript/jerry-core/vm/vm.c:5312
#14 0x00000000005ce740 in ecma_op_function_call_simple (func_obj_p=<optimized out>, this_binding=11, arguments_list_p=<optimized out>, arguments_list_len=<optimized out>)
    at /jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1180
#15 0x00000000005cdd10 in ecma_op_function_call (func_obj_p=func_obj_p@entry=0x1221508 <jerry_global_heap+776>, this_arg_value=this_arg_value@entry=771, 
    arguments_list_p=arguments_list_p@entry=0x7fffffffd440, arguments_list_len=arguments_list_len@entry=1) at /jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1414
#16 0x000000000060dffc in ecma_regexp_replace_helper (this_arg=3616307, string_arg=<optimized out>, replace_arg=<optimized out>) at /jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:2619
#17 0x00000000007518c8 in ecma_builtin_regexp_prototype_dispatch_routine (builtin_routine_id=<optimized out>, this_arg=771, arguments_list_p=<optimized out>, arguments_number=<optimized out>)
    at /jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-regexp-prototype.c:602
#18 0x00000000005809a5 in ecma_builtin_dispatch_routine (this_arg_value=771, arguments_list_p=0x6, arguments_list_len=1, func_obj_p=<optimized out>)
    at /jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1460
[... 19 lines cut ...]
    at /jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1379
#29 0x00000000006c370c in opfunc_call (frame_ctx_p=0x7fffffffdd00) at /jerryscript/jerry-core/vm/vm.c:758
#30 vm_execute (frame_ctx_p=frame_ctx_p@entry=0x7fffffffdd00) at /jerryscript/jerry-core/vm/vm.c:5217
#31 0x00000000006c0ce4 in vm_run (shared_p=shared_p@entry=0x7fffffffde40, this_binding_value=this_binding_value@entry=11, lex_env_p=lex_env_p@entry=0x12214e8 <jerry_global_heap+744>)
    at /jerryscript/jerry-core/vm/vm.c:5312
#32 0x00000000005ce740 in ecma_op_function_call_simple (func_obj_p=<optimized out>, this_binding=11, arguments_list_p=<optimized out>, arguments_list_len=<optimized out>)
    at /jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1180
#33 0x00000000005cdd10 in ecma_op_function_call (func_obj_p=0x12214d8 <jerry_global_heap+728>, this_arg_value=this_arg_value@entry=72, arguments_list_p=arguments_list_p@entry=0x7fffffffe0e8, 
    arguments_list_len=arguments_list_len@entry=1) at /jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1414
--Type <RET> for more, q to quit, c to continue without paging--
#34 0x00000000005cda16 in ecma_op_function_validated_call (callee=731, this_arg_value=72, arguments_list_p=arguments_list_p@entry=0x7fffffffe0e8, arguments_list_len=arguments_list_len@entry=1)
    at /jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1379
#35 0x00000000006c370c in opfunc_call (frame_ctx_p=0x7fffffffe0a0) at /jerryscript/jerry-core/vm/vm.c:758
#36 vm_execute (frame_ctx_p=frame_ctx_p@entry=0x7fffffffe0a0) at /jerryscript/jerry-core/vm/vm.c:5217
#37 0x00000000006c0ce4 in vm_run (shared_p=shared_p@entry=0x7fffffffe1e0, this_binding_value=<optimized out>, lex_env_p=0x12212f8 <jerry_global_heap+248>) at /jerryscript/jerry-core/vm/vm.c:5312
#38 0x00000000006c0607 in vm_run_global (bytecode_p=0x1221798 <jerry_global_heap+1432>, function_object_p=<optimized out>) at /jerryscript/jerry-core/vm/vm.c:286
#39 0x00000000004ed25d in jerry_run (script=script@entry=715) at /jerryscript/jerry-core/api/jerryscript.c:554
#40 0x0000000000804393 in jerryx_source_exec_script (path_p=<optimized out>) at /jerryscript/jerry-ext/util/sources.c:68
#41 0x00000000004e0161 in main (argc=<optimized out>, argv=<optimized out>) at /jerryscript/jerry-main/main-desktop.c:156

Debug:
Error: JERRY_FATAL_REF_COUNT_LIMIT
Aborted

```

closed 2024-08-14 by zherczeg

connglli, 2024-08-15:
Hi @kulcsaradam, I'm new to JerryScript but came across a similar issue recently. The issue only shows in the release version.  When building JerryScript with `--debug`, the program exits normally.

The issue within the release version has a similar stack trace to the above but I didn't notice any `JERRY_FATAL_REF_COUNT_LIMIT` things:

```
> lldb ./build/bin/jerry
(lldb) target create "./build/bin/jerry"
Current executable set to '/.../build/bin/jerry' (x86_64).
(lldb) run /.../test.js
Process 95932 launched: '/.../build/bin/jerry' (x86_64)
Process 95932 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007ff81380f7a2 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`:
->  0x7ff81380f7a2 <+10>: jae    0x7ff81380f7ac            ; <+20>
    0x7ff81380f7a4 <+12>: movq   %rax, %rdi
    0x7ff81380f7a7 <+15>: jmp    0x7ff813809184            ; cerror_nocancel
    0x7ff81380f7ac <+20>: retq   
Target 0: (jerry) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007ff81380f7a2 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007ff813847f30 libsystem_pthread.dylib`pthread_kill + 262
    frame #2: 0x00007ff813766a49 libsystem_c.dylib`abort + 126
    frame #3: 0x0000000100109b67 jerry`jerry_port_fatal(code=JERRY_FATAL_REF_COUNT_LIMIT) at jerry-port-process.c:29:5 [opt]
    frame #4: 0x0000000100098b44 jerry`jerry_fatal(code=JERRY_FATAL_REF_COUNT_LIMIT) at jrt-fatals.c:63:3 [opt]
    frame #5: 0x000000010000f4b4 jerry`ecma_ref_object_inline(object_p=<unavailable>) at ecma-gc.c:143:5 [opt]
    frame #6: 0x00000001000207ce jerry`ecma_copy_value(value=859) at ecma-helpers-value.c:894:7 [opt]
    frame #7: 0x000000010002082f jerry`ecma_fast_copy_value(value=<unavailable>) at ecma-helpers-value.c:921:76 [opt] [artificial]
    frame #8: 0x00000001000f552f jerry`vm_loop(frame_ctx_p=0x00007ff7bfefdfe0) at vm.c:1044:9 [opt]
    frame #9: 0x00000001000f3a28 jerry`vm_execute(frame_ctx_p=0x00007ff7bfefdfe0) at vm.c:5211:37 [opt]
    frame #10: 0x00000001000f2be5 jerry`vm_run(shared_p=0x00007ff7bfefe120, this_binding_value=11, lex_env_p=0x000000010014c130) at vm.c:5312:10 [opt]
    frame #11: 0x0000000100073303 jerry`ecma_op_function_call_simple(func_obj_p=<unavailable>, this_binding=11, arguments_list_p=<unavailable>, arguments_list_len=<unavailable>) at ecma-function-object.c:1180:28 [opt]
    frame #12: 0x0000000100072dea jerry`ecma_op_function_call(func_obj_p=0x00000001001c0748, this_arg_value=72, arguments_list_p=0x00007ff7bfefe3cc, arguments_list_len=3) at ecma-function-object.c:1414:16 [opt]
    frame #13: 0x0000000100072cd9 jerry`ecma_op_function_validated_call(callee=<unavailable>, this_arg_value=<unavailable>, arguments_list_p=<unavailable>, arguments_list_len=<unavailable>) at ecma-function-object.c:1379:10 [opt] [artificial]
    frame #14: 0x00000001000f3f19 jerry`vm_execute [inlined] opfunc_call(frame_ctx_p=0x00007ff7bfefe380) at vm.c:758:5 [opt]
    frame #15: 0x00000001000f3e43 jerry`vm_execute(frame_ctx_p=0x00007ff7bfefe380) at vm.c:5217:9 [opt]
    frame #16: 0x00000001000f2be5 jerry`vm_run(shared_p=0x00007ff7bfefe4c0, this_binding_value=11, lex_env_p=0x000000010014c130) at vm.c:5312:10 [opt]
    frame #17: 0x00000001000f295a jerry`vm_run_global(bytecode_p=0x000000010014c610, function_object_p=<unavailable>) at vm.c:286:25 [opt]
    frame #18: 0x0000000100005baa jerry`jerry_run(script=779) at jerryscript.c:554:24 [opt]
    frame #19: 0x0000000100108f1c jerry`jerryx_source_exec_script(path_p=<unavailable>) at sources.c:68:14 [opt]
    frame #20: 0x00000001000015d6 jerry`main(argc=<unavailable>, argv=<unavailable>) at main-desktop.c:156:20 [opt]
    frame #21: 0x00007ff8134bd386 dyld`start + 1942
(lldb) 
```

The code for test.js is:

```js
var v0 = [];
var v2 = new Int32Array();
for (var i = 0; i < 88815; i++) {
    async function f3(a4, a5, a6) {
        await a6;
        return a4;
    }
    f3(Int32Array, v2, v0);
}
```

**Note**: Even if I change the number of loop iterations to `8881500`, there's still no abortion in the debug version. So perhaps it is not an OOM?

Could you share some insights to this issue?

zherczeg, 2024-08-16:
`JERRY_FATAL_REF_COUNT_LIMIT` means there are too many references to one object. You should use 32 bit refcounts, instead of 16 bit.