jerryscript-project/jerryscript #5156: Why it runs failed to run a javascript code

RingsC, 2024-08-13:
I use the following compiling command to build the static libs, and use these libs in my project.
`
cmake .. -DCMAKE_BUILD_TYPE=Release -DJERRY_CMDLINE=OFF -DBUILD_SHARED_LIBS=OFF -DJERRY_ERROR_MESSAGES=ON -DCMAKE_INSTALL_PREFIX=/path/to/install
`
When i use `jerry_parse ` and `jerry_run` to compile the following part of javascript. but it failed. Can you tell me what's wrong with my code, and run the example code that it works well.  I just replace the script with my own code, and re-run the example , it does not work.

`
var VAL = 5; function isEven(num) {    return num % 2 == 0;} return isEven(VAL);
`

`
  const jerry_char_t script[] = "var VAL = 5; function isEven(num) {    return num % 2 == 0;} return isEven(VAL);";

  /* Setup Global scope code */
  jerry_value_t ret_val = jerry_eval (script, sizeof (script) - 1, JERRY_PARSE_NO_OPTS);

  print_value(ret_val);
  /* Returned value must be freed */
  jerry_value_free (ret_val);

`
and it saw the `ret_val` that it 's an object type.

zherczeg, 2024-08-13:
I suspect `jerry_init()` is missing, see here:

https://github.com/jerryscript-project/jerryscript/blob/master/tests/unit-core/test-abort.c#L38

RingsC, 2024-08-13:
> I suspect `jerry_init()` is missing, see here:
> 
> https://github.com/jerryscript-project/jerryscript/blob/master/tests/unit-core/test-abort.c#L38

At first, i think that i do the intialization in main thread, and run the script in another thread that leads to this error. But, i am wrong. After i do initialzation `jerry_init (JERRY_INIT_EMPTY);` in same place (same thread), it still gets an exception report.

RingsC, 2024-08-13:
And, i will build a debug version of jerryscript, which can step into the source code, to find out the reason.

zherczeg, 2024-08-14:
Is there anything else which run? I think `return` cannot be used in global code, only in a function.

RingsC, 2024-08-14:
> Is there anything else which run? I think `return` cannot be used in global code, only in a function.

Yes, you're right, `return` can not be used in global code.  It works without that `return`, and Thanks again for your help.

closed 2024-08-14 by RingsC